tscircuit/props #211: Add diode variants `variant="[zener, schottky, photo]"`

seveibar, 2025-04-26:
https://news.ycombinator.com/reply?id=43799609&goto=item%3Fid%3D43798832%2343799609

msmuenchen, 2025-04-26:
as mentioned in HN, I'd also add TVS diodes, they have their own symbols as well https://de.wikipedia.org/wiki/Suppressordiode

seveibar, 2025-05-04:
@msmuenchen ty! changed issue title to reflect.

Note to implementor:

- diode should accept `variant="schottky"`, zener, photo or tvs. We should also accept "standard" (which is a default)
- diode should have booleans for each type, i.e. `schottky: z.boolean().optional()`
- there should be a transform when parsing to set the booleans based on variant (but not vice versa). If no flags are set `standard: true`
- No two booleans can be set, otherwise error

Afterwards, the symbol should change accordingly, that implementation will be in https://github.com/tscircuit/core

CC @techmannih

seveibar, 2025-05-04:
/bounty $15

Anshgrover23, 2025-05-04:
/attempt #211

PR #216 (2025-05-04): Add diode variants variant=[zener, schottky, photo, tvs]
/claim #211
Fixes #211
closed 2025-05-04 by seveibar